traccar/traccar-client-ios #18: Traccar client issue

KumarK80, 2015-03-23:
Hi,

Thanks for giving us this great app. Traccar client hangs when I give my server ip address and I am not sure why.
1. I installed the client, did not make any changes to the server details - it worked fine - I was able to see the status and locate the position in your demo server.
2. I installed the server in my laptop and used ngrok for port forwarding and everything worked fine.
3. I installed the server in VPS and trying to update the new ip in the client - no success - the status screen not showing any status details, if I try to change the ip address, it hangs. Not sure what is the problem.

For all the 3 above I am using iPhone 6 plus.
Looking forward to hearing from you.

Thanks
Kumar

tananaev, 2015-03-23:
Check that the port is visible from Internet. For example [here](http://www.tananaev.com/port.html).

KumarK80, 2015-03-23:
Hi Tananaev,

Thank you very much. It was a port issue and when I opened the port, the
client is working fine.

However I am not able to find my device in the web now. In the log file
"tracker-server.log" I see an entry for my device '352714' but there is no
"tracker-server.log.yyyymmdd" file created.

Earlier I was trying to connect to MySQL and made some changes to the conf
file. From last 1 hour I am switching to and fro from H2 to SQL making
changes to the conf file hoping to see the device either in H2 or mysql.
Currently the entries in the conf file looks like below and not able to see
the device in web

<entry key='database.driver'>org.h2.Driver</entry>
<entry key='database.url'>jdbc:h2:/opt/traccar/database</entry>
<entry key='database.user'>sa</entry>
<entry key='database.password'></entry>

Can you please point out what mistake I might be making. Should I restart
the service to make the changes in conf file effective? Please let me know.

Thanks
Kumar

tananaev, 2015-03-23:
What do you see in the log? Does it say unknown device?

KumarK80, 2015-03-23:
Hi, Yes it says unknown device

tananaev, 2015-03-23:
It means that you haven't added your device to the database through web interface.

KumarK80, 2015-03-23:
I had added it earlier...and now I switched off and then switched on the
service status button and I was able to identify the device. Thank you for
your help and this great application.

tananaev, 2015-03-23:
Traccar Client sends identification only once right after establishing network connection, so you need to add id first and then click start service button.

KumarK80, 2015-03-25:
Hi Anton,
I am trying to move from H2 to MySQL. I created database traccar in database & made below changes in the conf file:
    <entry key='database.driver'>com.mysql.jdbc.Driver</entry>    <entry key='database.url'>jdbc:mysql://74.124.xxx.xx:3306/traccar?allowMultiQueries=true&amp;autoReconnect=true</entry>    <entry key='database.user'>test</entry>    <entry key='database.password'>password</entry>
I switched off and switched on traccar client. Then I logged out and closed the browser and then logged in again. 
Now I don't see the tables/ data in database. However if in browser I am still able to see the latest location. I guess still the H2 is being used. Can you please let me know what I need to do to get the data in MySQL?
ThanksKumar

tananaev, 2015-03-25:
You need restart Traccar server.

KumarK80, 2015-03-26:
Hi Anton,
I restarted the server and now I am getting Remote procedure call error. I went through some of the earlier threads and made sure that user have all the access rights. Tried installing latest version as well... Still getting the same error. Can you pls let me know how can I resolve this?
ThanksKumar

tananaev, 2015-03-26:
Check wrapper.log file for detailed error message.

KumarK80, 2015-03-27:
Hi Anton,

I tried to go through the wrapper.log and could not figure out what is the
problem (not an expert in Java). I have attached the wrapper and server log
file. Can you please help me in resolving this?

Thanks
Prasanna

tananaev, 2015-03-27:
You cannot attach files to GitHub emails. If you want you can send it to me by email (address in profile).

tananaev, 2015-03-28:
I see you already created another thread:
https://github.com/tananaev/traccar/issues/1143