traccar/traccar-client-ios #19: App tracking works on app store release but not on source code release

EdGoodall01, 2015-03-30:
Hi

Using the source code for iOS from the website, the location does not update. However, if I use the app store release, it does. Does anyone know why this is, or what I am doing wrong? I'm probably doing something wrong (more likely, I'm very tired!)

Thanks

tananaev, 2015-03-30:
You can try to compile code from the latest tag:
https://github.com/tananaev/traccar-client-ios/releases/tag/v1.4

EdGoodall01, 2015-03-31:
Tried using that code, still the issue occurs. It doesn't request access to the location if that helps? And in the status does not show any location failed or location sent...

tananaev, 2015-03-31:
It is the code that I used to build App Store version, so I don't understand how it's possible that App Store version works and your one doesn't.

App requests location in the following line, so you can try to put breakpoint and see if you it gets there:
https://github.com/tananaev/traccar-client-ios/blob/master/TraccarClient/ViewController.m#L93

EdGoodall01, 2015-03-31:
Taken a look, still no changes and the breakpoints aren't really revealing anything. Were there any frameworks you added or something like that? I'm quite confused here as obviously it is the app store code, but the location dialog doesn't appear at all asking if I would like to allow the app to use the location?

tananaev, 2015-03-31:
I think it won't appear if you disable locations in the settings.

EdGoodall01, 2015-03-31:
@tananaev tried with on and off, several devices and tried re-downloading code - definitely missing something on my end. Any suggestions as to things I could have missed myself?

tananaev, 2015-04-01:
Not sure what can go wrong there. The app is pretty simple. Maybe better to ask on StackOverflow because I'm not an iOS guru.

EdGoodall01, 2015-04-01:
Can you let me know what version of Xcode you were using incase the frameworks are different?

tananaev, 2015-04-01:
I think I uses Xcode version 5. Don't remember minor version number.

EdGoodall01, 2015-04-01:
@tananaev So glad you said that as that is what I suspected - I'll try with v5 and let you know as I think the location framework and syntax changed between 5 and 6 (or something like that...)

EdGoodall01, 2015-04-02:
Compiling with Xcode 5.1.1 works and submits location.

closed 2015-04-02 by EdGoodall01

tananaev, 2015-04-02:
I guess I need to update something if I want to release a new version. Thanks.