traccar/traccar-client-ios #20: iOS Version 1.5 Fails to post

netsound, 2015-06-05:
My iPhone6+ just updated to 1.5 traccar client and now it will not post location updates..  Service starts connection succeeded but then nota.. No positions posted...   Have removed and re-installed the app.. Also just tried on my backup iphone 4s and same issue.   Even tried the demo server.   Android app works fine though...

tananaev, 2015-06-05:
I have already fixed the issue and now waiting for Apple to review the app which takes about a week. So, the fixed version should be available on the App Store sometime next week.

The problem was that they changed the way location APIs work in iOS 8.

netsound, 2015-06-05:
Awesome...  Is there a way you could send me a .ipa to test.. We have a BikeMS event this weekend and worried some of the other iphones updated to 1.5 as well...  We have more android and GSM devices running but besides myself there are 3 other iOS devices running that I know of.....

By the way awesome job with everything on traccar, I setup the server about a month ago to replace a hacked togther setup of gpsgate.com server software we had been running for a year or so and so much more happier with traccar.  With putting the data right into mysql was able to write scripts to take the data and port it to APRS-IS so that it is combined with our VHF ax.25 based trackers as well..  Have run 3 different events with apps and GSM based trackers so far and no issues what so ever it just works and with less VM horse power then all the gpsgate.com servers we had running in the past..

tananaev, 2015-06-05:
I need Device IDs to make a build for you. More info on how to find IDs here:
https://developer.apple.com/library/ios/documentation/IDEs/Conceptual/AppDistributionGuide/MaintainingProfiles/MaintainingProfiles.html#//apple_ref/doc/uid/TP40012582-CH30-SW46.

netsound, 2015-06-05:
0a279c053289ebd228c35d5b5f303ac50dc6255c

If the others did update I will just bring my spare GSM trackers and do a simple pre-paid sim for the weekend..

tananaev, 2015-06-05:
Here is the IPA file for version 1.6:
https://www.dropbox.com/s/kp9x4xdhoutu0kv/TraccarClient.ipa?dl=0

netsound, 2015-06-12:
Sorry for the delay just getting back into town.... Worked perfectly.. Thanks again and for all the hard work you do on the project..

tananaev, 2015-06-12:
I got an email from Apple today that they have started review process, so hopefully app will be available on the App Store later today.