traccar/traccar-client-ios #23: The app don't transmit

maxlife70, 2015-09-06:
Good morning,
i have upgraded the last version of the app on my iphone 4.
But when i put the trigger on to start the transmission the app close and when I reopen it the trigger is off.
I'm using the traccar server version 2.11.
Thanks in advance for any helps.
Best Regards,

Max

tananaev, 2015-09-06:
It means that the app was kill either by user or the system. Did you kill it? If not, then most likely it was killed by the system because of low memory.

maxlife70, 2015-09-07:
Hi Tananaev,
thanks for your reply.
The app close itself, but if I double click on home button i found it in the app running and if i reopen the app the trigger is off.
Seems like the app is autominimized
I hope to be clear, my english not is very good.
BR

tananaev, 2015-09-07:
If the service status flag was reset to off, it means that `applicationWillTerminate` method was called by the system (app was killed). There is nothing I can do to prevent the OS from killing the app. Are you sure you have `Background App Refresh` enabled in the phone settings?

maxlife70, 2015-09-09:
Yes the Background App Refresh is enabled.
Thanks
BR

tananaev, 2015-09-09:
Don't have any other ideas. The only thing left to do is to get system logs from iPhone. Official tool to access console logs is Xcode, but there must be some third party apps for that as well.