traccar/traccar-client-ios #24: SOS button Developer

kochphilipp94, 2015-10-31:
Hello i search a developer what my can make a SOS button to the Traccar Client ( iOS / Android )

When i Press the Button then musst on the Web begin flashing....
Also i search people what my can make on the Web a special solution with push or IP call...

Please send my a msg when you can doing this to : chlinekoch@gmail.com

JurajBaran, 2015-11-02:
Hello, I have made text messages to Traccar, and the next update can be calls... but it is still work in progress.

kochphilipp94, 2015-11-02:
hello, have you becoming a date, when the new version coming ?

tananaev, 2015-11-02:
@kochphilipp94, are you asking about official Traccar version? @JurajBaran is talking about his own custom build as far as I can tell.

kochphilipp94, 2015-11-02:
@tananev i search a Solution for my problem ( official or custom ) can you my help by my problem ?

tananaev, 2015-11-02:
I was thinking of including the SOS button functionality into the app, but it might take a while till I find time to implement it.

kochphilipp94, 2015-11-02:
I have a old traccar server with the sos botton i the app , i will now find a person what my can update to the new traccar version and look thats then the SOS also works ..

tananaev, 2015-11-02:
Do you have some screenshots of what it looks like and how it works?

kochphilipp94, 2015-11-30:
@tananaev 
can you my give your email adress ? 
i will you send the login , this is better then the screenshots.

tananaev, 2015-11-30:
Email address is in the profile.