traccar/traccar-client-ios #25: Hello!    Traccar 2.0 on iOS 7.2.1 does not work(on2of iPhones4)-please help2downgrade2Traccar 1.6 which worked good.    Thanx in advance!

DevEroni, 2015-11-06:
Hello!

Traccar 2.0 on iOS 7.2.1 does not work(on2of iPhones4)-please help2downgrade2Traccar 1.6 which worked good.

Thanx in advance!

tananaev, 2015-11-06:
What port are you using?

tananaev, 2015-11-06:
Please don't write your question in the topic. There is a description field for it.

DevEroni, 2015-11-06:
Hello, Anton!

linux:50005 

but App brokes the same for  demo.traccar.org:5055 &  demo2.traccar.org:5055.

for andorid it works  well now

tananaev, 2015-11-06:
You should be using port 5055. What is the device id you tried on the demo server?

Anychka, 2015-11-06:
937918
as i see it start to connect,but fail - without records to log( - most of all, the reason is conflict between iOS/iPhone & App
'cause with traccar1.6 it worked well on my 50005 port.
so how can i downgrade?

PS if i've tried to start service right after install it changed

tananaev, 2015-11-06:
I don't see any data from your device in the server logs. What do you see on the status screen? (ideally attach a screenshot)

Anychka, 2015-11-06:
nothing new4u 2 show - right after starting service it's closing imidiatly and in couple seconds location icon at status bar disapear

tananaev, 2015-11-06:
Do you mean the app crashes? Why does it close by itself? If it does, I need crash logs.

Anychka, 2015-11-06:
the reason is upgrade 2 ver2.0 - checked it on two devices(

tananaev, 2015-11-06:
That's not the reason. I want to know why it's happening.

Anychka, 2015-11-06:
{"name":"TraccarClient","bug_type":"109","os_version":"iPhone OS 7.1.2 (11D257)","bundleID":"org.traccar.client.TraccarClient","version":"2.0 (2.0)","app_name":"TraccarClient"}
Incident Identifier: C430C2CE-9D17-4281-A447-289F055F982D
CrashReporter Key:   98d4487debe23186be6af846dad06fb265faf5f9
Hardware Model:      iPhone3,2
Process:             TraccarClient [192]
Path:                /var/mobile/Applications/7B4E7771-C248-4C5A-BE4E-657482EA1DB2/TraccarClient.app/TraccarClient
Identifier:          org.traccar.client.TraccarClient
Version:             2.0 (2.0)
Code Type:           ARM (Native)
Parent Process:      launchd [1]

Date/Time:           2015-11-07 01:27:13.221 +0200
OS Version:          iOS 7.1.2 (11D257)
Report Version:      104

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000
Triggered by Thread:  0

Last Exception Backtrace:
(0x2f287f7e 0x39a0bcca 0x2f1c1f08 0x65a0e 0x67e9e 0x67dcc 0x67d4a 0x67c10 0x645e8 0x2f74a874 0x2f746a1e 0x2f73ffbe 0x2f253234 0x2f2525b6 0x2f251224 0x2f1bbeba 0x2f1bbc9e 0x340c165e 0x31b08148 0x649fe 0x39f18ab2)

Thread 0 Crashed:
0   libsystem_kernel.dylib        	0x39fce1f0 0x39fbb000 + 78320
1   libsystem_pthread.dylib       	0x3a038792 0x3a035000 + 14226
2   libsystem_c.dylib             	0x39f7efd8 0x39f36000 + 298968
3   libc++abi.dylib               	0x392adcd2 0x392ad000 + 3282
4   libc++abi.dylib               	0x392c66e0 0x392ad000 + 104160
5   libobjc.A.dylib               	0x39a0bf62 0x39a08000 + 16226
6   libc++abi.dylib               	0x392c41c4 0x392ad000 + 94660
7   libc++abi.dylib               	0x392c3d28 0x392ad000 + 93480
8   libobjc.A.dylib               	0x39a0be12 0x39a08000 + 15890
9   CoreFoundation                	0x2f1bbf30 0x2f1b4000 + 32560
10  CoreFoundation                	0x2f1bbc9e 0x2f1b4000 + 31902
11  GraphicsServices              	0x340c165e 0x340b8000 + 38494
12  UIKit                         	0x31b08148 0x31a9a000 + 450888
13  TraccarClient                 	0x000649fe 0x5c000 + 35326
14  libdyld.dylib                 	0x39f18ab4 0x39f17000 + 6836

Thread 1:
0   libsystem_kernel.dylib        	0x39fbb808 0x39fbb000 + 2056
1   libdispatch.dylib             	0x39f07de8 0x39ef2000 + 89576
2   libdispatch.dylib             	0x39ef6f6e 0x39ef2000 + 20334

Thread 2:
0   libsystem_kernel.dylib        	0x39fbbac0 0x39fbb000 + 2752
1   libdispatch.dylib             	0x39f06f40 0x39ef2000 + 85824
2   CoreLocation                  	0x2f73fea0 0x2f73d000 + 11936
3   CoreLocation                  	0x2f744b04 0x2f73d000 + 31492
4   CoreLocation                  	0x2f7725b0 0x2f73d000 + 218544
5   libxpc.dylib                  	0x3a045382 0x3a041000 + 17282
6   libxpc.dylib                  	0x3a043936 0x3a041000 + 10550
7   libdispatch.dylib             	0x39ef4582 0x39ef2000 + 9602
8   libdispatch.dylib             	0x39ef3d9e 0x39ef2000 + 7582
9   libdispatch.dylib             	0x39ef4352 0x39ef2000 + 9042
10  libdispatch.dylib             	0x39ef3d9e 0x39ef2000 + 7582
11  libdispatch.dylib             	0x39ef4292 0x39ef2000 + 8850
12  libdispatch.dylib             	0x39ef3d9e 0x39ef2000 + 7582
13  libdispatch.dylib             	0x39ef4292 0x39ef2000 + 8850
14  libdispatch.dylib             	0x39f0688a 0x39ef2000 + 84106
15  libdispatch.dylib             	0x39f06b1c 0x39ef2000 + 84764
16  libsystem_pthread.dylib       	0x3a035bd0 0x3a035000 + 3024
17  libsystem_pthread.dylib       	0x3a035a94 0x3a035000 + 2708

Thread 3:
0   libsystem_kernel.dylib        	0x39fcec70 0x39fbb000 + 81008
1   libsystem_pthread.dylib       	0x3a035bda 0x3a035000 + 3034
2   libsystem_pthread.dylib       	0x3a035a94 0x3a035000 + 2708

Thread 4:
[... 189 lines cut ...]
0x39eb2000 - 0x39eb8fff libcopyfile.dylib armv7  <236a4387651534e9a9dd670ca9db02bb> /usr/lib/system/libcopyfile.dylib
0x39eb9000 - 0x39ef1fff libcorecrypto.dylib armv7  <912e56f196943500877e1aba46efcb78> /usr/lib/system/libcorecrypto.dylib
0x39ef2000 - 0x39f16fff libdispatch.dylib armv7  <ffb7ebb4a206398fbf7507e05e6b7b96> /usr/lib/system/libdispatch.dylib
0x39f17000 - 0x39f18fff libdyld.dylib armv7  <e3df65531c733096b90b3187b7894592> /usr/lib/system/libdyld.dylib
0x39f19000 - 0x39f19fff libkeymgr.dylib armv7  <a671e49149e7326dafa58332efdb4bd0> /usr/lib/system/libkeymgr.dylib
0x39f1a000 - 0x39f20fff liblaunch.dylib armv7  <1efc4b3c146d347cb586fce80e8acdcb> /usr/lib/system/liblaunch.dylib
0x39f21000 - 0x39f24fff libmacho.dylib armv7  <0ffeb9459e8d3ba99467a159b8c23d78> /usr/lib/system/libmacho.dylib
0x39f25000 - 0x39f26fff libremovefile.dylib armv7  <ab36d6cb22b833e9bce2c2c375f57cea> /usr/lib/system/libremovefile.dylib
0x39f27000 - 0x39f34fff libsystem_asl.dylib armv7  <6569ced4c68b3a119b89f3a8f48564be> /usr/lib/system/libsystem_asl.dylib
0x39f35000 - 0x39f35fff libsystem_blocks.dylib armv7  <0178b6f1a9bf362d9672641076a42ee2> /usr/lib/system/libsystem_blocks.dylib
0x39f36000 - 0x39f97fff libsystem_c.dylib armv7  <371f0e1854f03b05ba143619e4eac99f> /usr/lib/system/libsystem_c.dylib
0x39f98000 - 0x39f9afff libsystem_configuration.dylib armv7  <3e82bcb352f53951b5bce186bf9c30f7> /usr/lib/system/libsystem_configuration.dylib
0x39f9b000 - 0x39fa1fff libsystem_dnssd.dylib armv7  <b80a9e53918437fba1e1db7ba356a378> /usr/lib/system/libsystem_dnssd.dylib
0x39fa2000 - 0x39fbafff libsystem_info.dylib armv7  <2996bfe2d4833c20b428a1b7b4db3c52> /usr/lib/system/libsystem_info.dylib
[... cut ...]
0x3a035000 - 0x3a03afff libsystem_pthread.dylib armv7  <bdb1cf2e67983b65b5bd694b3cde3dc0> /usr/lib/system/libsystem_pthread.dylib
0x3a03b000 - 0x3a03cfff libsystem_sandbox.dylib armv7  <a4289fa3fcb03507bcc49b3ea68b6e86> /usr/lib/system/libsystem_sandbox.dylib
0x3a03d000 - 0x3a03ffff libsystem_stats.dylib armv7  <0f72cd8b79ad3600b2eb01c4391b2f53> /usr/lib/system/libsystem_stats.dylib
0x3a040000 - 0x3a040fff libunwind.dylib armv7  <45c29718368e3d1c8821911ab5953080> /usr/lib/system/libunwind.dylib
0x3a041000 - 0x3a05bfff libxpc.dylib armv7  <399deef84c9e3570a39520230faa3216> /usr/lib/system/libxpc.dylib

tananaev, 2015-11-07:
Symbolicated stack trace:

<pre>+[TCProtocolFormatter formatPostion:address:port:] (in TraccarClient) (TCProtocolFormatter.m:29)
-[TCTrackingController read] (in TraccarClient) (TCTrackingController.m:117)
-[TCTrackingController write:] (in TraccarClient) (TCTrackingController.m:107)
-[TCTrackingController didUpdatePosition:] (in TraccarClient) (TCTrackingController.m:86)
-[TCPositionProvider locationManager:didUpdateLocations:] (in TraccarClient) (TCPositionProvider.m:78)</pre>

tananaev, 2015-11-07:
It looks like something is wrong with `deviceId`. Try to remove documents and data for the Traccar Client app and try again.

traccar2, 2015-11-07:
i've installed it on other dev (without old app tried) - app there also crashed & closed by itself(

tananaev, 2015-11-07:
Have you changed any settings? Can I see a screenshot of the main screen?

traccar2, 2015-11-07:
[image]

traccar2, 2015-11-07:
that's all? nothing else matter?

tananaev, 2015-11-07:
I tried it on 8.4 simulator and everything works fine. I will download and try version 7 as well.

traccar2, 2015-11-07:
эмуляция это здорово, но если на двух айфонах не работает, то неужели дело не очевидное и нет возможности для 7ки ставить версию2 ?

tananaev, 2015-11-07:
Дело в том что я за больше чем два месяца с момента выпуска 2-ой версии я ни разу не слышал ни про какие проблема. Это при том что приложение установлено на тысячах телефонов.

traccar2, 2015-11-07:
Антон, сомневаюсь что пользователей Traccar'а на ios7 много и того меньше, кто будет разбираться - работает хорошо, не работает - ну и х... с ней, этож ябло-пользователи, а не дроидоводы...
да и пусть будет как есть, хотя раньше в АппСторе была опция для  иОС6 устанавливать устаревшие, но совместимые именно с этой версией Апп"ы
прошу лишь помощи в установке старой версии 1.6 или саму - просто у меня аналогичных тел у родственников 5шт...
PS Auckland в Калифорнии или новой зеландии, если CA , то GMT-8 ? - могу голосом на брать на +1559, а то и заехать - расскажете за Питер)

tananaev, 2015-11-07:
Попробую завтра разобраться с проблемой.

Auckland в Новой Зеландии.

closed 2015-11-08 by tananaev

tananaev, 2015-11-08:
Исправил пробему. Осталось дождаться когда Apple опубликует новую версию приложения.

traccar2, 2015-11-08:
Anton, thanks alot. Can it be done via dropbox ?

tananaev, 2015-11-08:
Unfortunately no. Apple needs to review and approve it before anybody can install it on their devices.

commit f9717c5d (2015-11-08): Fix crash on iOS 7 devices (fix #25)