traccar/traccar-client-ios #27: Iphone Send Failed

pedrovasques77, 2015-11-11:
I am trying to connect my iphone to the demo server and i get send failed.
Can someone help me?

tananaev, 2015-11-11:
What port are you using?

pedrovasques77, 2015-11-11:
I am using the connection details above:

IP: 104.43.219.242
Port: 5005

tananaev, 2015-11-11:
You need to use port 5055.

pedrovasques77, 2015-11-11:
Thanks for your quick response.
I changed the port but still "Send Failed" on status

tananaev, 2015-11-11:
What is the device id?

pedrovasques77, 2015-11-11:
Thanks for your help. now it is working as expected.

tananaev, 2015-11-13:
I've just received a notification from Apple that Traccar Client version 2.1 has successfully passed the review and now available on the App Store. Please let me know if everything works correctly in the new version.

traccar2, 2015-11-14:
Hello, Anton!
Traccar Client version 2.1 works even on iPhone4(hw3.2). 
It reports data in 5minutes with frequency 60 setuped.
But there lots of "send failed" errors in app log.
So app retry 2send data every minutes. That discharges accum very fast.
ids are:
00 or 000 or 5 on demo2
&
5 on demo

tananaev, 2015-11-14:
Main demo server has been unstable recently. Can you please try `demo2.traccar.org` and see if problem persist.

traccar2, 2015-11-14:
>00 or 000 or 5 on demo2

tananaev, 2015-11-14:
On the screenshot it was still `demo.traccar.org`.

I can see only one message from "000" and two messages from "5".

traccar2, 2015-11-14:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

tananaev, 2015-11-14:
Can you please try `dev.traccar.org`.

traccar2, 2015-11-14:
Антон, на демо серверах обновлялось ПО до 3.2 ? на Дев"е предыдущая 3.1 ?

tananaev, 2015-11-14:
Да, я обновил сервера. На dev все работает?

traccar2, 2015-11-14:
[image]
[image]

tananaev, 2015-11-14:
Все доходит на сколько я вижу из логов:

<pre>WARN: Unknown device - 000
WARN: Unknown device - 400612
WARN: Unknown device - 5</pre>

Я заметил интересный момент. Идентификатор 400612 это случайно не ваш старый идентификатор? Штука в том что старые данные буферизируются (включая идентификатор) в базе если нету соединения и отправляются позже. Это может объяснить все "Send failed".

traccar2, 2015-11-14:
не попали в http://dev.traccar.org/
все верно 400612 мой первичный
Вам виднее - напрягает что их по два фэйла в минуту и как следствие быстрый разряд акума(
увеличение частоты с 60 до 600 не влияет(

tananaev, 2015-11-14:
Не попали потому что в базу не заведены.

traccar2, 2015-11-14:
Какие еще будут предложения по отладке?

tananaev, 2015-11-14:
Да ничего отлаживать не нужно. Заведите все идентификаторы (включая старые) на сайт и все заработает.

traccar2, 2015-11-14:
Речь о дев-сервере?

tananaev, 2015-11-14:
О любом сервере. Все идентификаторы должны быть заведены.

traccar2, 2015-11-15:
Антон, заработало-спасибо! 
планируется ли в ябло-версии скрытая версия как у андроида и 
функционал как в апк Валерия Калашникова, 
например, отсыл данных только в случае изменения?

tananaev, 2015-11-15:
Скрытой версии думаю что не будет потому что на iOS нету технической возможности такое реализовать на сколько я знаю.

Отправка данные по расстоянию а не по времени планируется.

marialin, 2017-01-23:
I am trying to connect iphone 6 plus to demo2 server. I get the send failed message. I am using the demo2.traccar.org, The server address is 149.56.170.158, and server port is 5055. How to solve the problem?

marialin, 2017-01-23:
My device id is 235231.

tananaev, 2017-01-23:
@marialin, I don't see any data from your device. It probably means that the settings are incorrect or there is some network issue on your phone.

marialin, 2017-01-23:
I restarted my iPhone and now I can see myself on the map after login to demo2.traccar.org. I can see myself walking around my office :). Thanks.

albita10, 2019-06-28:
I am trying to connect my iPhone 7 plus and I receive the failed shipping message.My server addres is 190.167.121.76 port is 5055. How to solve the problem?

tananaev, 2019-06-28:
@albita10 follow troubleshooting guide. I you still have questions, ask on forum, not here.

albita10, 2019-06-28:
Still giving me error

albita10, 2019-06-28:
I followed the steps of the instructions and is still giving me the shipping error.

Sirokoff78, 2020-11-12:
Добрый день. Установил Traccar Client 6.1 на iphone 6s (iOS 14.2). Приложение запускается, но позицию не передает. Пишет в состояние: service created ... и все.
В настройках iOS, в приложения отсутствует пункт разрешение использования геопозиции.
В самом приложении настройки не соответствуют приложенному в appstore скриншоту. Например нет поля "порт".