traccar/traccar-client-ios #28: Number of devices

pedrovasques77, 2015-11-17:
I am using traccar only with a few devices and i have a server dedicated to this.
What is your experience about the number of devices per server instance we can track simultaneously?

tananaev, 2015-11-17:
I know some customers have 10k+ real devices working with Traccar. Also Traccar has been successfully tested with 30k simulated devices.

pedrovasques77, 2015-11-17:
30000 devices real time on one single server?
what number of lines you will have on database?

tananaev, 2015-11-17:
I didn't do the test. Someone else did.

I can give you some statistics around demo server. It has a few thousands of registered devices. Not sure how many of them are online at the moment. Database has only around 10 million records because I clean it regularly.

pedrovasques77, 2015-11-17:
If possible send me the hardware details you use for the demo server

tananaev, 2015-11-17:
It uses Azure cloud service. 4 cores and 7GB RAM.