traccar/traccar-client-ios #31: iOS Battery Level always 0

bradleyuk, 2015-12-24:
Hi

Not sure if this a defect or future proofed but the latest iOS client always seems to send batt=0 at the end of the query string. 

Is this a placeholder for when battery level is supported or is this a defect?

Also quick question - if i increase the interval will it save battery? or is the GPS always on (and therefore draining) and just transmits at the interval provided?

Thanks

And great app btw!

tananaev, 2015-12-24:
Looks like I need to set `batteryMonitoringEnabled` property to `YES` to be able to read battery status.

PoD-Stas, 2016-06-22:
Simple solution is:

`-(double) batteryLevel
{
    UIDevice *myDevice = [UIDevice currentDevice];
    [myDevice setBatteryMonitoringEnabled:YES];
    double batteryLeft = (float)[myDevice batteryLevel] * 100;
    return batteryLeft;
}`

tananaev, 2016-06-22:
I will try to include in the next release.

PoD-Stas, 2016-06-22:
I did like it in my branch. Work fine but mistake near +/- 5-10%

w32gold, 2017-01-05:
Hi, used the latest Traccar-iOS codes but still getting the "zero battery". Any further ideas in fixing this? thanks.

tananaev, 2017-01-05:
It means that your device doesn't provide this information.

w32gold, 2017-01-08:
Sorry for my ignorance, but how do I make the iPhone/iPad provide such battery information.
I've seen it sending a battery percentage before. Thanks!

tananaev, 2017-01-15:
The issue should be fixed now. We need to wait a bit till Apple approves the new version.

commit dd78d173 (2017-01-15): Read real battery level (fix #31)
closed 2017-01-15 by tananaev