traccar/traccar-client-ios #35: Auto start service

PoD-Stas, 2016-03-04:
Hi!
I have jailbreaked devices. and I need to automatically start service.
Don't u know how can I modify this app, or make some tweak for cydia to do that on my device? :)
Or maybe I can modify deployed app to do that?

tananaev, 2016-03-04:
You can use `voip` background mode. Here is a quote from Apple documentation:
> The app provides Voice-over-IP services. Apps with this key are automatically launched after system boot so that the app can reestablish VoIP services. Apps with this key are also allowed to play background audio.

tacman, 2016-03-17:
Is this the only way to do that?

tananaev, 2016-03-17:
Yes, it is the only way.

tacman, 2016-03-17:
How about significant location change?  That's also a valid background / auto-launch, right?

tananaev, 2016-03-17:
Background location updates mode doesn't auto-start an app.

tacman, 2016-03-17:
If the app is running in the background and the phone is rebooted, you have
to manually re-open the app and start the service?  I thought in the past
this ran continuously.   Am I misunderstanding something?

tananaev, 2016-03-17:
Yes, you have to re-open the app. Only Traccar Client for Android has auto-start option.

PoD-Stas, 2016-04-12:
So the only reason is Apple politics about privacy? Technically for personal use do I just have to enable "voip" background mode to the [main plist](https://developer.apple.com/library/ios/documentation/Performance/Conceptual/EnergyGuide-iOS/Art/xcode_project_capabilities_backgroundmodes_2x.png) to add?
And then launch
`self.trackingController = [[TCTrackingController alloc] init];
        [self.trackingController start];`

From for example `- (void)viewDidAppear`

https://developer.apple.com/library/ios/documentation/Performance/Conceptual/EnergyGuide-iOS/Art/xcode_project_capabilities_backgroundmodes_2x.png

tananaev, 2016-04-12:
I guess you can do it for your own developer build, but it will never be approved for the App Store.

PoD-Stas, 2016-04-12:
Okay thanks! Yes it's just for my personal use.

jeanlk15, 2022-06-21:
can I ask if the auto start of traccar is still an issue on iphones ?