traccar/traccar-client-ios #36: iOS traccar hidden just like Android

kkar, 2016-04-07:
Hello again,
is there a way to hide the app in an iOS device without jailbreak?

I also found traccar client in windows store for windows based phones but i am unable to find any git about it. Is there any / is there a hidden version of it for windows phones?

Thanks again for your support.

tananaev, 2016-04-07:
No, as far as I know there is no way to hide an app on iOS device.

App in the Windows store is not an official Traccar project. I plan to develop Windows client in future.

wolfbgn, 2016-05-14:
Isn't it possible just to remove the little arrow next to battery indicator, which appear when Trracar client is ON?
Now if the phone is stolen, the thief can see immediately that some tracking program is working.

PoD-Stas, 2016-05-14:
No it isn't, due iOS limitation and apple policy. Using any GPS software on iOS will cause that indicator appeared.
To prevent stealing just use find my iPhone app.