traccar/traccar-client-ios #38: iOS client build

wolfbgn, 2016-04-29:
Hello, Mr. Tananaev.
What program should I use to build from source this application? I would like to change IP address and port. I am asking this question assuming that it is similar to Android apk building in Android Studio.
I use Windows 10 and Linux ( Debian ). 
Thank you.

PoD-Stas, 2016-04-29:
Hello! I'm not mr. Tamara.
But let me answer for him. Well it's iOS app so u can build it only in Xcode which available under Mac OS. (OS X).
There no other ways.
For example u can run os x under virtual machine.

But to change server and port u can just type it after app is launched :D

tananaev, 2016-04-29:
Comment from @PoD-Stas is correct. You need Mac OS and Xcode to build iOS app. There is no way around it.

PoD-Stas, 2016-04-30:
@tananaev in prev reply iphone did an autocorrection on ur name, sry))

wolfbgn, 2016-04-30:
Thank you very much to both of you. 
I hoped there is some other way since I have no experience with Mac OS.