traccar/traccar-client-ios #39: IOS app sleep mode

JohnnyDevine, 2016-10-02:
Would it be possible to add a sleep mode to the IOS app so if the phone is stationary/inactive then the  app is not draining the battery, with the app set to 60 seconds, my battery was nearly flat overnight.

tananaev, 2016-10-02:
Unfortunately it's not possible. Apple is very restrictive with what app can do in the background.

JohnnyDevine, 2016-10-02:
I think I had an IOS app that did this a few weeks back Anton, would you like me to backtrack if I can remember?

tananaev, 2016-10-02:
Well, as far as I know, it's not possible. If you can find a way, I'm definitely interested.

JohnnyDevine, 2016-10-02:
ah sorry Anton, i got the threads crossed, i thought you were responding to the remote on/off request

JD