traccar/traccar-client-ios #41: App outside the Apple Store

Turbovix, 2016-12-06:
Hello,
Is there any way to get the app to install via Cydia? Because Apple will not let me install the Traccar on an Iphone 4, since the IOS of it is 7.xxx

tananaev, 2016-12-06:
It has nothing to do with Apple. The app project is built with 8.0 as a target SDK, so it won't install on anything older.

You can try to build it yourself with lower SDK, but some things might not work or you might get some compilation issues.

Turbovix, 2016-12-06:
Youtube was not available, however, I did install an old version through Cydia, and then the app was updated by the apple store itself.
Thank you.

tananaev, 2016-12-07:
You can always get an older version tag from the repository and compile it yourself.

Turbovix, 2016-12-09:
OK thank you!
It worked very well!

closed 2017-09-17 by tananaev