traccar/traccar-client-ios #42: How to get user location when app is killed/terminated in background state

kolindalanaresh, 2017-02-08:
how to get user location when app is terminated/killed by the user at background state.

tananaev, 2017-02-08:
Obviously app can't do anything if it's terminated. Please clarify the question.

kolindalanaresh, 2017-02-09:
traccar app is working in background and if it is terminated from background user location not getting according to apple there is an option startmonitoringsignificantlocationchanges.
where to use this startmonitoringsignificantlocationchanges in traccar.

tananaev, 2017-02-09:
Well, the `startMonitoringSignificantLocationChanges` method might actually do the trick. I would have to investigate it more and check where we can call it.

Knowledgetreedata, 2018-01-17:
Hi , Same issue have added in startMonitoringSignificantLocationChanges set always and allowed background refresh and set app as priority but still termination of connection even though IOS Traccar Client On Iphone 6, 6s and 7 90% devices with one device remaining constantly on. Is there specific iOS version issue or relates to iOS functionality over-riding app. Please help

Knowledgetreedata, 2018-01-17:
PS if you have direct email I can provide log in email and password to perform programme check and am happy to pay. 60% of driver offline due to iOS

tananaev, 2018-01-17:
Email is public in my profile and in repo README file.

Knowledgetreedata, 2018-01-23:
Hi anton

Just a quick question tried to find a guide for data use assuming 60 seconds using iphone and android phone is australia. Do you have a sample sheet or guide for data usage. I know its dependent upon telecom provider but what data size is transferred as an estimate per 60 seconds

tananaev, 2018-01-23:
There were some estimates on the forum. FIY, you are emailing to github thread.