traccar/traccar-client-ios #44: Changing Project Name and other naming conventions in Xcode 8

dpepperd, 2017-04-16:
I've spent hours mostly looking on stackoverflow and apple which were mostly old and not accurate. I eventually found a video on YouTube that got me almost where I was trying to get to. Changing the project name and the way Xcode deals with recursive directory structures is not as easy as it sounds. 

Here's a link to the video that helped me:

https://www.youtube.com/watch?v=jRnVjtNLLLk

Hopefully the last issue that I'm having trouble resolving is a file not found error with:

_InAppSettingsKit/IASKAppSettingsViewController.h_

If someone can help me figure this out I would greatly appreciate it. I'm pretty sure if I can get this header fixed it will compile correctly.

Thanks!

Dirk Pepperd

tananaev, 2017-04-16:
Are you using latest source code? Just earlier today I've updated project to use Carthage for dependencies management.

dpepperd, 2017-04-16:
Yes I just downloaded it again from github a few hours ago Anton. Thanks if
you can help me figure it out!

tananaev, 2017-04-16:
Install carthage and run following command:
```
carthage update --platform iOS
```

Alternatively you just use a version from tag which doesn't require any extra steps:

https://github.com/tananaev/traccar-client-ios/tree/v4.0

dpepperd, 2017-04-16:
Thanks Anton! Carthage helped fix the issue. I appreciate you getting right
back to me!

dpepperd, 2017-07-27:
Anton,

I am working with a Tkstar tracker that Kevin Stevens from Vitaltrace UK
discovered. He has it running on port 5013. I'm trying to get messages to
the panel but so far nothing is showing up in the log files. Could it be
that you built the protocol for that into his panel? Let me know i'm
interested in getting this device working. Hope all is well with you.

Thanks,

Dirk

tananaev, 2017-07-27:
How is this question related to Traccar Client for iOS? I'm not sure if you realise that you posted a comment on GitHub.