traccar/traccar-client-ios #47: Multiple server reporting

webby52, 2017-05-20:
Is there a way to add additional servers to push gps data to? I need to send the data to two different servers in the iOS app.

tananaev, 2017-05-20:
It's not possible out of the box. You would have to make some code modifications if you want to achieve it.

webby52, 2017-05-20:
Great. How hard is it? Can I just embed it into the code and not build a form for it?

Will Mullen
Deputy Chief 52
Hopewell Fire Dept
4 Columbia Ave
Hopewell, NJ 08525
609.466.0060 (fh)
609.577.5050 (c)
609.466.8590 (f)

tananaev, 2017-05-20:
Of course you can. It's easy if you have iOS development experience.

webby52, 2017-05-21:
Yea. Not so much. 

Would there also be a way to make it stop sending gps when the iPad stopped moving, after a predetermined amount of time? Then when it starts moving again resume sending? 

I am using this in a fire truck with an iPad with a cellular connection. 

Is there a price for you to customize to our need? 

Will Mullen
Deputy Chief 52
Hopewell Fire Dept
4 Columbia Ave
Hopewell, NJ 08525
609.466.0060 (fh)
609.577.5050 (c)
609.466.8590 (f)

tananaev, 2017-05-21:
In the latest version of Traccar Client you can set up reporting by distance, so it won't send if it's not moving. It will still drain battery by using GPS though.

We can help with customization. Send me an email to discuss it. Address is in profile.