traccar/traccar-client-ios #48: iOS Client Updating but not showing records on Traccar Server

Sharingan86, 2017-06-05:
Hi Anton,

We are running Traccar 3.7 and have recently tried to get my iPhone 6s Plus (iOS 10.3.2) to send records in using the Traccar Client app. So far the client is updating location but no records are showing up on the server.

I have service status enabled and Encryption disabled for this test, I have tried with encryption on also.

tananaev, 2017-06-05:
What do you mean by "client is updating location"?

Sharingan86, 2017-06-05:
The Traccar Client app is showing that it is updating location.
[image]

tananaev, 2017-06-05:
Check this guide:

https://www.traccar.org/troubleshooting/