traccar/traccar-client-ios #52: Idea for battery saving

tananaev, 2017-06-09:
Run low accuracy location updates, so update is not called every second. Then in callback request high accuracy location.

Suggested by Apple engineer.

tananaev, 2017-11-25:
Implemented location accuracy option for now instead.

closed 2017-11-25 by tananaev

LNFWebsite, 2017-12-10:
Hi, thanks for making such a useful GPS tracking service!

Traccar Client works great, except for the immense battery usage even at slow intervals on iOS.
I was curious as to when you'll be able to push this change to the App Store?

Thanks ahead.

tananaev, 2017-12-10:
Probably in a couple of weeks time.

LNFWebsite, 2017-12-12:
Okay, thanks for the quick response!

LNFWebsite, 2018-02-22:
Any update as to the battery drain issue?

tananaev, 2018-02-22:
The app has been released. What kind of updates are you looking for?

LNFWebsite, 2018-02-22:
I'm sorry, I hadn't noticed that you updated the app...

I'll try it out again and let you know whether the battery drain is any better.

tananaev, 2018-02-22:
Make sure you select medium or low accuracy.

LNFWebsite, 2018-02-23:
Yes, the battery drain is much less! Thank you very much!