traccar/traccar-client-ios #54: Geolocation acting strange

simplywing, 2017-07-06:
I recognized it a few times now that the first one or two location points are very far off the real point.  After that it seems to recover. (Screenshot) This only seems to happen on IOS and not on Android.
[image]

tananaev, 2017-07-06:
Possibly it's network location, which can be very inaccurate. Then GPS fixes the signal, so you start receiving precise location.

simplywing, 2017-07-06:
Is there a way to fix this and only take the GPS-Location?

tananaev, 2017-07-06:
Unfortunately not for iOS, but it should be reporting accuracy along with location.

simplywing, 2017-08-09:
Is there a documentation available for the accuracy reporting?

According to the protocol formater, the accuracy is not reportet:

`NSMutableString *query = [[NSMutableString alloc] init];
    [query appendFormat:@"id=%@&", position.deviceId];
    [query appendFormat:@"timestamp=%ld&", (long) [position.time timeIntervalSince1970]];
    [query appendFormat:@"lat=%.06f&", position.latitude];
    [query appendFormat:@"lon=%.06f&", position.longitude];
    [query appendFormat:@"speed=%g&", position.speed];
    [query appendFormat:@"bearing=%g&", position.course];
    [query appendFormat:@"altitude=%g&", position.altitude];
    [query appendFormat:@"batt=%g", position.battery];
    components.query = query; // use queryItems for iOS 8`

https://github.com/tananaev/traccar-client-ios/blob/master/TraccarClient/TCProtocolFormatter.m

tananaev, 2017-08-09:
Looks like it's not implemented on iOS yet. Android app definitely reports it.

simplywing, 2017-08-10:
https://github.com/tananaev/traccar-client-android/blob/master/app/src/main/java/org/traccar/client/ProtocolFormatter.java

It does not seem to...

tananaev, 2017-08-10:
Maybe I have some changes locally. I definitely remember implementing it. Anyway, it's in plans to include it.

simplywing, 2018-05-07:
Hi, I revisited this topic since the Android now nicely reports the location accuracy. Cool! Is it planned to be implemented in IOS soon?

tananaev, 2018-05-07:
It should be available in iOS as well.

simplywing, 2018-05-08:
It clearly is not: 
https://github.com/traccar/traccar-client-ios/blob/master/TraccarClient/ProtocolFormatter.swift

I used Wireshark to be sure. It really does not report the accuracy, no matter which accuracy mode is chosen.

tananaev, 2018-05-08:
I definitely worked on it. Possibly I haven't pushed it to the GitHub yet for some reason.

simplywing, 2018-05-08:
Would be super nice if you could push these changes.

LNFWebsite, 2018-07-13:
@tananaev Thanks for working on the ability to change accuracy so as not to drain battery.

Unfortuately though, the accuracy radius is not being reported when in low accuracy mode. Is this known behavior of the app?

Thanks ahead!

tananaev, 2018-07-13:
I don't think we released a new version with this functionality yet.

LNFWebsite, 2018-07-14:
Oh okay. Yeah, an accuracy radius would be really helpful information if it's possible to gather. In the low accuracy mode, you can't distinguish between super inaccurate locations by cell triangulation from locations gathered by WiFi or GPS.

Thanks anyway.

tananaev, 2018-07-14:
It will be included with next release.

LNFWebsite, 2018-07-24:
Okay, thanks man!