traccar/traccar-client-ios #58: Version 5.0 problems

newbiemmxvii, 2017-09-16:
Version 4.2 worked like a charm.
Version 5.0 doesn't work anymore.
Re-installs didn't help either.
Battery is above 90%.
Default settings used, except Traccar server
After install the dialog to grant permission to access GPS shows up immediately but too short to be answered.
Needed to use privacy settings to permit Traccar client to access GPS
Service stops right after the phone goes to sleep, i.e. requiring the PIN to re-gain access.
The fact that the service stopped is not shown (updated) in the app settings dialog (looks like service is still active). Deactivating and activating the service makes the game to start over.
iPhone 6s, iOS 10.3.3

tananaev, 2017-09-16:
Can you get system logs from your phone?

newbiemmxvii, 2017-09-16:
Will try, a hint how to get it would be appreciated, e.g. a link with es description

tananaev, 2017-09-16:
Maybe this would help:

https://www.theiphonewiki.com/wiki/System_Log

newbiemmxvii, 2017-09-16:
Thanks, will look into it and share what I find

newbiemmxvii, 2017-09-16:
Please check your traccar eMail inbox.

tananaev, 2017-09-16:
I have checked logs and looks like app gets suspended:
```
Sep 16 17:24:56 iPhone mediaserverd(CoreMedia)[27] <Notice>: -CMSessionMgr- cmsmHandleApplicationStateChange: CMSession: Client org.traccar.client.TraccarClient with pid '443' is now Background Suspended. Background entitlement: NO
Sep 16 17:24:56 iPhone mediaserverd(AudioToolbox)[27] <Notice>: 1130: pid 443(TraccarClient)
Sep 16 17:24:56 iPhone mediaserverd(AudioToolbox)[27] <Notice>: 3147: sid:0x1b198, TraccarClient(443), 'prim' new state = Background Task Suspended
```

There are also some network issues:
```
Sep 16 17:25:02 iPhone TraccarClient(libsystem_network.dylib)[443] <Error>: nw_socket_get_input_frames recvmsg(fd 12, 1024 bytes): [57] Socket is not connected
...
Sep 16 17:25:02 iPhone TraccarClient(CFNetwork)[443] <Error>: TIC Read Error [4:0x170189720]: 1:57
Sep 16 17:25:02 iPhone TraccarClient(CFNetwork)[443] <Error>: TIC Read Error [4:0x170189720]: 1:57
```

Unfortunately it doesn't really help much to understand what's happening.

I am also investigating some crash issues. Maybe it's related to your problem.

newbiemmxvii, 2017-09-17:
Thank you! 

Why does the first log entry (you referenced) state: _Background entitlement: NO_

Regarding networking I would want to understand what the traccar client is trying to achieve. The firewall might be an issue if dedicated (nonstandard) ports are used.

tananaev, 2017-09-17:
I'm not really sure what those logs mean. I was also a bit surprised by no background entitlement, but maybe it supposed to work like that.

I have uploaded a new version to the App Store. Hopefully it solves some problems. It's waiting for Apple review and should be available in a couple of days.

newbiemmxvii, 2017-09-17:
I would doubt that might be supposed to work like that. The semantics of _Background entitlement: NO_ makes me believe that the app is not supposed to keep running in the background.

This might give some insight:

https://developer.apple.com/library/content/documentation/iPhone/Conceptual/iPhoneOSProgrammingGuide/BackgroundExecution/BackgroundExecution.html

Sections _Declaring Your App’s Supported Background Tasks_ and  _Tracking the User’s Location_

My apologies if the link references well known basic stuff

tananaev, 2017-09-17:
Of course it's well known. We use `location` option from there. I guess log is for other type of background entitlement.

newbiemmxvii, 2017-09-18:
Update:

 An iPhone 5s is affected identically by the same problem .

Hint:
Tracking is NOT suspended when the status page is kept open, i.e. the position update log entries can be observed before entering idle mode.
Tracking is suspended when the Traccar Client tab (with all configuration options) is open and the phone goes into idle mode.

tananaev, 2017-09-19:
I have updated the app. Please try version 5.1. It is waiting for Apple review and should be available in a couple of days if everything goes well.

mwl, 2017-09-19:
I have the exact same problem on two separate devices (iOS 10.3.3 and iOS 11) running Traccar Client 5.0 with the only exception that I can't even allow the app to see the location in Settings.app. These are completely new installations and have never been hooked up with any Traccar server before.

tananaev, 2017-09-19:
Please wait and see if it works on Traccar Client 5.0.

mwl, 2017-09-22:
Just installed 5.1, and I'm afraid is has changed nothing. The iOS Location permissions dialog comes up for a second or so, before it disappears. In status it says
> Service created
> Service destroyed

I even did a clean install.

newbiemmxvii, 2017-09-22:
Acknowledgement, no change with 5.1, the problem persists, sorry.
Have to keep it running with the status page compulsory open before it changes to background mode to prevent of "Service destroyed".

tananaev, 2017-09-22:
Can anyone please send a video of the problem?

tananaev, 2017-09-22:
Also, please anyone who can help with testing through TestFlight, please send me your email addresses. You can send them to my email address (in profile).

mwl, 2017-09-23:
Sure, here you go https://youtu.be/09ftaTCLPHk

Sorry I didn’t change the system language, but the pop up dialog is asking for permission to use my location. I’m not responding on it and it goes away by itself.

tananaev, 2017-09-23:
Please wait for version 5.2. It should definitely be fixed now.

In the meanwhile what you can do is manually enable location permissions for the app from settings.

mwl, 2017-09-23:
Thanks Anton. I can confirm the workaround works.

dpepperd, 2017-09-25:
In regard to the popup dialog, I ended up going into settings and just changing the permission there to allow traccar client to use location services. I have not been able to get version 5.1 to run on Ios 11 on the new Iphone 8 at all. Anton if there is anything you need to help you out post it here I'll be glad to do whatever I can.

newbiemmxvii, 2017-09-26:
Thank you Anton! 5.2 sounds promising.

tananaev, 2017-09-26:
It should already be available. Please let me know if it works.

mwl, 2017-09-26:
Sorry to report, that the issue is exactly the same on 5.2.

Another thing I've noticed with 5.1 over the weekend was that tracking only works with you leave the app in the Status view.

dpepperd, 2017-09-26:
Same issues with 5.2 on iOS 11 and iPhone 8. App is not working.

tananaev, 2017-09-26:
OK, I believe at least one of the problems should be fixed now. Permissions dialog should function correctly now. Please confirm.

As for the other problem, it's a mystery. I have no idea why it works if you open status screen but not on the main screen. For now my suspicion is that Apple did some "optimization" which causes this behavior. Is the behavior same for everyone? How long does app report with locked screen? Does it stop reporting instantly?

mwl, 2017-09-26:
> OK, I believe at least one of the problems should be fixed now. Permissions dialog should function correctly now. Please confirm.

It is _still_ the same issue. Dialog shows up for a second, then disappears again by itself.

tananaev, 2017-09-26:
Are you sure you are using version 5.2? I definitely fixed that problem. Tested on iPhone 7.

dpepperd, 2017-09-26:
Issue still exists on ios 11 and iphone 8. Dialog box pops up and goes away
quickly.

mwl, 2017-09-26:
> Are you sure you are using version 5.2? I definitely fixed that problem. Tested on iPhone 7.

For as long as I deleted the app and reinstalled it through the App Store after confirming that 5.2 was available. So pretty sure, yes :)

pmlewisuk, 2017-09-26:
Hi

Iphone 6 IOS 10

With version 5.1 I found the service was destroyed when I exited the app, if I left it on the status screen and exited it was OK 

I updated to 5.2 and it was always immediately destroying so I uninstalled and then did a fresh 5.2 install.
Now the Allow 'Traccar Client' to Access flashes up and disappears, if I do manage to say allow I get service destroyed. It doesn't remember the 'Allow' and doesn't show in the privacy location services since the reinstall.

Hope this help you.

Regards

Mark

tananaev, 2017-09-26:
Really hard to debug issues without real iPhone device on hands. Anyway, I think I have fixed (again) the problem with disappearing dialog. Please check version 5.3 which should be available in a couple of days.

This time I tested on Amazon device farm on iPhone 5s with iOS 10.3.3.

dpepperd, 2017-09-26:
Anton,

If you want help testing I have an iPhone 8 and would be glad to help you.

ashishnaicker, 2017-09-27:
Same issue on iPhone 6 Plus, iOS 10.3.3, Traccar Client 5.2
- allow location dialog opens and closes almost immediately
- service starts and destroyed almost immediately
- device appears offline on demo server

wouterbudding, 2017-09-27:
I have the same on an iPhone 8 and on an iPhone 7, both with iOS 11 and iOS 11.0.1 , both Version 5.2 of the app. Please see attachment. aangemaakt = created and gestopt = stopped
[image]

tananaev, 2017-09-27:
Please check version 5.3. it's already under review, so hopefully it will be available in a few hours.

tananaev, 2017-09-27:
I just got notification that version 5.3 is out. Please let me know if it fixes the issues.

dpepperd, 2017-09-27:
I just went in App Store and 5.2 still showing up 9/27 at 19:23 EST.  I
will keep checking thanks.

pmlewisuk, 2017-09-28:
Just installed and tested, looking good 😊 

Will keep testing today

Thanks

newbiemmxvii, 2017-09-29:
Thank you Anton! Works fine for me. ***** The service doesn't stop anymore when the app is sent to background without showing the status page. I didn't test the dialog to grant GPS access though as I would have to reinstall the app and reenter the configuration, sorry. I trust it works.

This ticket could be closed in my eyes. Thank you again!

tananaev, 2017-09-29:
Great, thanks everyone for feedback.

closed 2017-09-29 by newbiemmxvii