traccar/traccar-client-ios #59: Does not buffer data for wifi if cellular data is turned off.

NoMoreNicksLeft, 2017-09-19:
If cellular data is turned on, I get beautiful tracking from this app. However, I'm not on an unlimited data plan, and while it's not especially verbose I tried to turn it off for the app (I still have a connection).

When in this mode, it seems to drop everything up until I get to home or work and have a wifi connection again.

This doesn't seem to be intended behavior that I have been able to determine, but possibly a side effect of iOS's ability to deny cell data to specific apps.

Though I have not had a chance to test it personally, I expect that it would work fine for tracking in places where cell coverage is nil.

tananaev, 2017-09-19:
It should save data. Are you sure it's not just a consequence of app not running in the background? See this:

https://github.com/tananaev/traccar-client-ios/issues/58

NoMoreNicksLeft, 2017-09-19:
I do not believe it to be so, because until I turned off the cell data, it was working well. I will run an experiment tomorrow to check for certain.

tananaev, 2017-09-19:
I would recommend to wait for version 5.1. It fixes some issues that are present in 5.0, so it might solve your problem as well.

mwl, 2017-09-22:
I bet there's a long waiting list in the App Store at the moment :)
If you fancy, I'm happy to test the app through TestFligt?

tananaev, 2017-09-22:
It's already picked for review, so hopefully it should be available in a few hours. If they find some problems we can try test flight.

mwl, 2017-09-22:
Just installed 5.1, and I'm afraid is has changed nothing. The iOS Location permissions dialog comes up for a second or so, before it disappears. In status it says
> Service created
> Service destroyed

I even did a clean install.

NoMoreNicksLeft, 2017-09-26:
I seem to have been mistaken with my initial assumption. The latest update seems to have fixed whatever actually was happening.

closed 2017-09-26 by NoMoreNicksLeft