traccar/traccar-client-ios #62: Lone worker features

markmcnamee, 2017-11-05:
As per forum post https://www.traccar.org/forums/topic/lone-worker-features/#post-22345

I think Traccar would really benefit from "Lone Worker Protection" features, primarily:

1) Heartbeat / check-in - the user must send an "ok" every x minutes or hours
2) SOS feature - would link in well with the existing SOS features but from the app
3) Fall detection mode - switchable. e.g. if you are working at height, enable the feature which could use the accelerometer in the phone

Use cases:

1) Worker in a remote location or isolated part of a larger building/factory/site, has to check in every x minutes or hours to ensure they are safe, awake and alive!

2) In the event of an emergency, the worker feeds under threat or in imminent danger, they can raise an SOS alarm via the app to be picked up by a remote user.

3) If the worker is going up a ladder or into a more dangerous environment, or a user is continually working in hazardous conditions (e.g. a warehouse forklift truck driver), they should enable the "Man down" feature. If the device accelerometer detects a sudden movement, followed by no movement for 1-2 minutes, or if they detect zero movement for 5-10 minutes plus, the phone should raise a man-down alert to the central server for the event to be actioned.

tananaev, 2017-11-06:
Traccar Client is really intended to be more generic app. I think the only generic feature from the list is SOS and it's already available in Android, so it makes sense to implement it. Other features seem to be very use case specific, but let's see if anyone else is interested in them as well.

meganhomekind, 2018-02-27:
I would be interested in the Heartbeat/check-in feature, as well as the SOS feature for iOS.

keshu0009, 2019-11-18:
I would be interested in the Lone worker features as well. Especially Fall Detection and Non-Movement Alert. Is Traccar in process of implementing the same?