traccar/traccar-client-ios #63: No Speed /  Heading

chasgames, 2018-02-05:
Not sure whether related to my iPhone X, but i'm experiencing these problems on the iOS Client:

- Only reports location if App is running in the background (I.E app has been clicked on since reboot, and not closed) If i remove it from recently opened apps, then it stops reporting my location. Should it not be reporting data entirely in the background?

- No Speed / Heading data being sent. Only Location.

tananaev, 2018-02-05:
1. When you remove app from background, system just kills it. It means nothing would work after that.

2. It means that you are not using GPS location (high accuracy).

For questions like this you should be using forum.

closed 2018-02-05 by tananaev

chasgames, 2018-02-05:
Wow that was a quick close!

1. I thought other apps like IFTTT can get my location whilst the app is closed using Background App Refresh/ "Long-Running Background Tasks". I don't see Traccar in the Background App Refresh list.

2. I've switched to high accuracy before (after searching around), I've done it again and still no speed or heading and I'm checking the latest logs and Course is always N in reporting. I've tried restarting the traccar client aswell. Still want me to go to the forums?

tananaev, 2018-02-06:
1. It's only possible for significant location updates (not realtime GPS) and for apps that were not explicitly stopped by user (your case).

2. Even with high accuracy you might not be using GPS. For example, GPS doesn't work indoors, so you get network location without speed and heading.

Yes, please for general questions use forum. Github is only for development-related discussions.