traccar/traccar-client-ios #68: I have a problem when I open the Safary explorer .

tocasoft, 2019-01-07:
Hello Anthon
I have a problem when I open the Safary explorer to show the window of the devices, reports and states. When I click on any of them, the tab does not open. you have some suggestion that you can help me. In windows it works well.
Thank you

tananaev, 2019-01-07:
How is this related to Traccar Client?

tocasoft, 2019-01-07:
Manager from web explorer in Safari.thanks

tananaev, 2019-01-07:
I'm closing this ticket because it clearly has nothing to do with Traccar Client app.

closed 2019-01-07 by tananaev

tocasoft, 2019-01-07:
How i can fix this problem my friend or I must open  other ticket?.thanks

tananaev, 2019-01-07:
You need to ask in appropriate place. If you think there is a bug in Traccar Manager, open ticket there. If you think it's web app issue, open ticket in web app repo.

tocasoft, 2019-01-07:
Ok. Thanks