traccar/traccar-client-ios #70: Feature Request

mdtaylorlrim, 2019-04-29:
Our non-profit organization utilizes traccar extensively, but we collaborate with other organizations that use traccar as well. It would be useful if traccar clients could store multiple server settings. At first a simple drop down list of servers and maybe later multiple profiles so that all settings could be unique to servers.

I wish I could contribute myself, I just don't have any IOS products other than my phone.

Thanks

tananaev, 2019-04-29:
You can configure Traccar server to forward data to a second server.

mdtaylorlrim, 2019-04-29:
Thank you Anton. Yes, and we do that in some cases. In one case the server is on a private network and has no access to the internet, although the private network is reachable countywide, but only if the internet is not available. Meaning, when the internet is down we can configure our clients to access the server over the private network, but the other server is not available. Therefore forwarding from server to server is not an option. I will have to be done at the client level.