traccar/traccar-client-ios #72: iOS Shortcuts and Automation

drguitarum2005, 2020-10-29:
Would it be possible to allow support by the iOS Shortcuts app, specifically automation routines? I'd like to be able to toggle the client on/off based on Geolocation (i.e. turn Traccar off if I'm going outside a certain zone) OR based on a schedule (i.e. turn Traccar off after 9 PM and back on at 6 AM).

tananaev, 2020-10-29:
Do you have any documentation on how to integrate it?

drguitarum2005, 2020-10-29:
Honestly, no. I'm not an iOS developer so I'm unsure how that part works. I just know when I attempt to make an automation, there are certain apps I can choose actions for and Traccar Client is not one of them. 

A more direct way would be adding the option right into the client. 2 fields for "start time" and "stop time" would be as simple as it would need to be.

tananaev, 2020-10-29:
I don't think adding it directly into the client is an option. We want to keep it simple, but it would be nice if there's a way to integrate it into the standard shortcuts app.

dfcowan, 2023-04-04:
I would like this feature also. The shortcuts already created for the `UIApplicationShortcutItems` could be adapted into intents or activities usable by the siri shortcuts system. Personally I would like the feature so that I can start/stop the service using a shortcut when airplane mode is toggled. 

https://developer.apple.com/documentation/sirikit/offering_actions_in_the_shortcuts_app

junglegamelab, 2023-09-03:
+1

Wanna turn on and off based on my location

joonhyuk, 2023-09-03:
+1

Wanna turn on and off based on home wifi connection

JoramHhub, 2024-03-08:
Would love to see this get implemented so I can start the service when my phone connects to my car’s bluetooth.

abdulmalekc1, 2024-10-07:
Feature implemented here #82 but the project seems abandoned

tananaev, 2024-10-07:
It's not abandoned.

Egregius, 2025-05-07:
Hi all,

Also posting this here because the pull request where I posted is closed and therefor may not be viewed.

I'm a new Traccar Client user. Formerly been using Owntracks to track my car trips but that one has no offline support and I already missed quit some tracks because of that.
I would like to start the Traccar Client's recording by using Siri Shortcuts when my phone gets connected to Carplay and stop it when disconnected. The automations are in place and working.
Problem is that in the Traccar Clients status windows only "Service created" and "Service destroyed" appear, no location update.
When I toggle the Server status switch off/on there's also instantly a location update.
So, when activated manually everything's working fine, when activated with the Siri shortcuts automation it doesn't.
Any ideas?