traccar/traccar-client-ios #76: Lost of log-in status and push notifications (Official iOS app)

castillo92, 2022-07-21:
**Describe the bug**
After updating from v4.9 to v5.2 and start using modern UI, I can not keep the login status in the mobiles app. Each start of app requires sign-in again. 

This affect to push notifications because since update I loose all of them except if you open the app and inmidiately trigger a "Test message" notification.

**Steps to reproduce the behavior:**
1. In the phone: Open official iOS app v2.10
2. In a computer: click on "Test message", you receive in the mobile one test notification: push and web.
3. In the phone: Minimize the app and wait 30 minutes... or more
4. Click again on "Test message", you will not receive any test notification: push or web.

**Expected behavior**
The app should keep open the login status and receive notifications despite you open or close the app

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version:  103.0.5060.134

**Smartphone (please complete the following information):**
 - Device: iPhone 13 and some Android
 - OS: iOS 15.5
 - Browser: Safari
 - Version: stock

**Additional context**
I came from Traccar Blog: https://www.traccar.org/forums/topic/traccar-manager-ios-app-lost-login-info-after-a-couple-of-minutes/#post-73265

Original issue in a bad repo: https://github.com/traccar/traccar/issues/4903#issue-1312965904

tananaev, 2022-07-23:
Just tested. After waiting 3 hours I still receive push notifications without any issues.

tananaev, 2022-07-23:
Fixed the login issue here: https://github.com/traccar/traccar-manager-ios/commit/011ddb9fcd1dfea68e8261bb2cc8b628c6175978.

By the way, this is also a wrong place for the ticket. Your report is about Traccar Manager, not Traccar Client.

closed 2022-07-23 by tananaev

castillo92, 2022-07-28:
> Fixed the login issue here: [traccar/traccar-manager-ios@011ddb9](https://github.com/traccar/traccar-manager-ios/commit/011ddb9fcd1dfea68e8261bb2cc8b628c6175978).
> 
> By the way, this is also a wrong place for the ticket. Your report is about Traccar Manager, not Traccar Client.

Thanks Antón, the main issue was the lost the login status. I am happy to know you found the fault. It will came in a iOS app update or is in the server side?

tananaev, 2022-07-28:
We need to update Traccar first because there are some new requirements from Apple.