traccar/traccar-client-ios #77: The project has compilation errors in the latest xcode14

gpsim, 2023-02-25:
**Describe the bug**
The project has compilation errors in the latest xcode14：
/traccar-client-ios/TraccarClient/AppDelegate.swift Could not build Objective-C module 'SwiftOverlayShims'

/traccar-client-ios/Pods/Firebase/CoreOnly/Sources/module.modulemap:1:8 Redefinition of module 'Firebase'

**To Reproduce**
Steps to reproduce the behavior:
1 Download the latest code through git

2. No changes have been made and cannot be compiled and run

**Expected behavior**
It may be a compatibility problem, but I don't know how to solve it. I hope you can help me analyze it. Thank you.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
- xcode Version 14.2 (14C18)
**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone11]
 - OS: [e.g. iOS16.3]

tananaev, 2023-02-25:
Are you sure you did a clean check of the latest code?

gpsim, 2023-02-26:
My mac has two accounts. Just switch to another account. It's strange. I'm very sorry for the confusion. It has been solved at present.

closed 2023-02-26 by gpsim