traccar/traccar-client-ios #78: Can't stay alive in background

bradley-rose, 2023-05-28:
**Describe the bug**
The app can't stay alive in the background. It stops logging points when not in the foreground. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open app
2. Start logging
3. Swipe up to put app in background

**Expected behavior**
Log points to Traccar Server

**Smartphone (please complete the following information):**
 - Device: iPhone 13 Mini
 - OS: 16.5

**Additional context**
There's no options for Background App Refresh for Traccar Client. It's just not listed in the list.

tananaev, 2023-05-28:
Have you checked system logs to see what happens?

PS Background App Refresh is something unrelated, afaik. We request background location updates instead.

bradley-rose, 2023-05-28:
> Have you checked system logs to see what happens?
> 
> 
> 
> PS Background App Refresh is something unrelated, afaik. We request background location updates instead.

Which system logs specifically? Somewhere in the iOS settings app that I should be looking?

tananaev, 2023-05-28:
iOS system logs.

bradley-rose, 2023-05-28:
There's no logs for Traccar Client. 

[image]

tananaev, 2023-05-28:
What is this? Once again, I'm talking about system logs. There should be a single log, not per app log.

bradley-rose, 2023-05-28:
Looks like the system log is only able to be accessed via desktop based on what I'm reading on a web search. I'll update later when I get around to investigating further.

bradley-rose, 2023-06-12:
Seems to have fixed itself over time. No direct actions were taken. Tested in outdoor, indoor, moving, and static environments, all seems to just work.

closed 2023-06-12 by bradley-rose