traccar/traccar-client-ios #80: Turn off tracking on closing app

Usernameclk, 2024-02-06:
Would it be possible to enable the app to turn off tracking when you come out of the app. Currently the only way to see if you have left the tracking on is to reopen the app and it immediately starts tracking you and announcing your location.
I don’t always remember to physically turn off service status when I am exiting the app and I am uncomfortable that it tells colleagues across the uk my location when I open the app to check it is turned off! 

Big fan of the app though - incredibly useful.

 **Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

tananaev, 2024-02-06:
If you close the app, it will stop tracking. If you minimize the app, it won't.

Usernameclk, 2024-02-06:
If you minimize the app it is still tracking in the background.. I understand that so I can use Traccar client in the background whilst using other apps. 

if you  swipe the app off, ) rather than physically going in to the app and turning off the service status) the service status is still on but it stops tracking. It only starts tracking again when you reopen the app. Strava, for example stops tracking you completely if you swipe the app off the screen. I was just wondering if there was a way to set it so that if I swipe the app off the screen it automatically turns off my service status and it stops tracking me.

tananaev, 2024-02-06:
If you know how Strava is doing it, we can try implementing it. I have a suspicion they might be using push notifications, which is against Apple policy.

Usernameclk, 2024-02-09:
I’m not any kind of IT expert so I have asked a friend… I hope his reply makes sense to you, not sure I fully understand his terminology! 

‘’I would agree that generally you would want to stop, or at least pause the tracking when the app closes. The developer should be able to subscribe to a system event to enable this.
That being said, they might want to add a feature for when you restart the app to ask the user if they want to continue tracking or not.
If they don’t get an event when the app closes, they certainly get one when the app is opened. At that point, if the tracking was still enabled, they should be able to stop it before it sends any location data.’’