traccar/traccar-client-ios #85: Add Support for Apples new "Background App Refresh"

Neon-44, 2024-12-15:
**Is your feature request related to a problem? Please describe.**
I just can't get Traccar to reliably update the location from iOS (It always kills the process). 

**Describe the solution you'd like**
Adding Support for "Background App refresh" would allow the App to continue sending the Location even when it's killed in the foreground. 

**Describe alternatives you've considered**
nothing

**Additional context**
on iOS: Settings -> General -> Background App Refresh

tananaev, 2024-12-15:
We already support background location, so I'm not exactly sure what the request is.

Neon-44, 2024-12-15:
Huh, it doesn't show up in the settings for me.

A Bug Report then?

tananaev, 2024-12-15:
Background Location Updates and Background App Refresh are two separate mechanisms in iOS.

Neon-44, 2024-12-15:
so, if I close the app (swipe it up) and the process gets killed that's expected behaviour?

If yes, would it be possible to implement background app refresh so the process stays alive?

tananaev, 2024-12-15:
If you explicitly kill the app (swipe it up), neither of those will work. If you want the app to keep running in the background, don't kill it.

closed 2024-12-15 by tananaev