traccar/traccar-client-ios #89: When setting 0 meter for update, no interval setting show up

tomlaf, 2025-06-14:
**Describe the bug**

On iOS when setting 0 meter to force continuous update on interval, the interval option does not show up, like the android app does.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to change setting
2. Set distance to 0
3. Interval does not show up

**Expected behavior**
No more update happens

tananaev, 2025-06-14:
It is expected. iOS only supports distance-based tracking.

closed 2025-06-14 by tananaev

tananaev, 2025-06-15:
You should check logs to see what happens. But in general Apple is much more restrictive with what we can do in the background, so I won't be surprised if it still kills the app.

tomlaf, 2025-06-15:
I just sent you the log.
I do not see the HearthBeat but I do see the preventSuspend task every minutes.
On the server I have activated database.saveEmpty to make sure I not missing anything, but the device still show as off line.

tananaev, 2025-06-15:
I recommend waiting for version 9.3.0.

tananaev, 2025-06-15:
They actually just approved it. Might still need to wait because the rollout takes some time.

tomlaf, 2025-06-15:
Thanks you for the update I will let know you the result of my test.

THOMAS-LOUIS LAFOREST
C : 1-819-580-0102